loboris/MicroPython_K210_LoBo #3: boot from SD card

Qutix, 2019-07-16:
Hello!

I built this repo on an Ubuntu and flashed it to my dev board:
[image]

However I have difficulties booting from SD card.
On every boot I have to use Putty.exe to connect to the board and execute the following code:
```
import os
os.mount(os.VfsSDCard(), '/sd')
os.chdir('/sd')
import boot
```

Could this be automated somehow? How can I check if an SD card is actually inserted?
An editable _boot.py file would be great like [this one](https://github.com/sipeed/MaixPy/blob/master/components/micropython/port/builtin_py/_boot.py)

I digged a little deeper and I found the [static const char* DEFAULT_BOOT_PY](https://github.com/loboris/MicroPython_K210_LoBo/blob/master/k210-freertos/main.c) but editing the string with newlines and space indents is a bit tedious. Is there a better option? Should I edit that line? If so what would be the best script? Should I just put the os.mount(...) in a try catch block and if there is an exception print out something along the line of "failed to mount SD Card"?

Side note:
I've checked the wiki: https://github.com/loboris/MicroPython_K210_LoBo/wiki/i2c
But for master I2C I got errors when I tried to use those arguments. For example "i2c.writeto(addr, buf [,stop=True])" has no stop argument. If I put True or False at the end I get argument related error messages. What am I doing wrong? Do I have to enable a flag in [mpconfigport.h](https://github.com/loboris/MicroPython_K210_LoBo/blob/master/k210-freertos/mpy_support/mpconfigport.h) file to have all the features? I see that there is a MICROPY_PY_MACHINE_I2C option disabled by deafult. But if I enable it nothing really changes. The output file has the sime size, same argument error. Maybe I mixed up something?

Qutix, 2019-07-16:
On my dev board I actually have a pin connected to the SD card socket's card detect pin:
[image]

But I understand that this pin is not used/connected on the original Sipeed boards so this might be a custom thing to check/have and might not have a future in the code.

loboris, 2019-07-19:
> However I have difficulties booting from SD card.
> On every boot I have to use Putty.exe to connect to the board and execute the following code:
> 
> ```
> import os
> os.mount(os.VfsSDCard(), '/sd')
> os.chdir('/sd')
> import boot
> ```
> 
> Could this be automated somehow? How can I check if an SD card is actually inserted?
> Side note:
> I've checked the wiki: https://github.com/loboris/MicroPython_K210_LoBo/wiki/i2c
> But for master I2C I got errors when I tried to use those arguments. For example "i2c.writeto(addr, buf [,stop=True])" has no stop argument.

If you want to mount SD Card on boot, just insert the code into `boot.py` on internal file system (`/flash).
If the SD Card is not inserted , the exception will be raised:
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 22] EINVAL
```
so it is best to insert this part of the code into `try - except` block.

There is no `stop` argument in any of I2C methods. I have fixed the Wiki, thanks for reporting this (The methods syntax was copied from my MicroPython for ESP32).

You don't need to use `Putty` to connect to the board.
`MPyTerm.py` terminal emulator is included which was specially created for this MicroPython portand has capability to transfer the files to/from the board.

Sorry for the uncomplete documentation, it will be updated and completed soon.

Qutix, 2019-07-27:
I've modified my /flash/boot.py like this:

`# This file is executed on every boot
import sys, os
sys.path().append('/flash/lib')

try:
    os.mount(os.VfsSDCard(), '/sd')
    os.chdir('/sd')
    sys.path.append('/sd')
    print('SD card successfully mounted.')
except:
    print('Could not mount SD card.')

devices = os.listdir('/')
if 'sd' in devices:
    if 'boot.py' in os.listdir('/sd'):
        print('boot.py found on SD card. About to start it...')
        import boot
    else:
        print('Could not find boot.py on SD card.')
`

It's working perfectly, however when I try to edit files on the SD card using the pye('/sd/boot,py') editor it saves all files with a wierd .pyetmp name. For example:
`from pye import pye
pye('/sd/boot2.py')
<add some code>
<press Ctrl + S>
for name it says boot2.py.pyetmp
<I delete the wierd ending and press Enter>
<press Ctrl + Q>
os.listdir('/sd')
boot2.py.pyetemp`

Can this be turned off somewhere?

loboris, 2019-07-27:
> ... when I try to edit files on the SD card using the pye('/sd/boot.py') editor it saves all files with a wierd .pyetmp at the end of file name...:

I couldn't reproduce this. The file is saved under requested name.
When saving the file **pye** first saves it with `.pyetemp` extension added, then renames it to the requested name.
```python
## write file
    def put_file(self, fname):
        tmpfile = fname + ".pyetmp"
        f = open(tmpfile, "w")
        for l in self.content:
            if self.write_tabs == 'y':
                f.write(self.packtabs(l) + '\n')
            else:
                f.write(l + '\n')
        f.close()
        try:
            is_file = os.check_file(fname)
            if is_file == 2:
                os.remove(fname)
            os.rename(tmpfile, fname)
        except:
            pass
```
If the code in `try ... except` block raises an exception, the saved file will not be renamed.

Qutix, 2019-07-27:
I did try to do the renaming and it gave me an error. Let me reproduce it and attach the error message here. 

```MaixPy-FreeRTOS by LoBo v1.11.6
-------------------------------
MicroPython 1.11.6 (6f4a861-dirty) built on 2019-07-27; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>> os.listdir()
['boot.py.pyetmp', 'boot2.py.pyetmp', 'System Volume Information', 'H\u0151- \xe9s \xe1rmal\xe1stan 2015-16 2 f\xe9l\xe9v 1_el\u0151ad\xe1s.pdf', 'power.py', 'display.py', 'led.py']
>>> os.rename('boot.by.pyetmp', 'boot.py')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
`

Qutix, 2019-07-27:
If I do

```
os.rename('/sd/boot.py.pyetmp', 'boot.py')
```

Then it works. Maybe there is a pathing issue?

loboris, 2019-07-27:
Yes, I've just discovered it myself. I'll try to fix it...

Qutix, 2019-07-27:
If I do
```from pye import pye
pye('/sd/boot.py')
```
Then it saves without any problem.

loboris, 2019-07-27:
**This bug is now fixed.**

Thank you for noticing and reporting the issue.
It looks I've never tested renaming the file in the current directory.

closed 2019-07-28 by Qutix