loboris/MicroPython_K210_LoBo #7: WiFi firmware

nalzok, 2019-08-04:
After reading the document I'm not quite clear about how to get Wi-Fi work on a Maix M1w. Here are some elementary questions:

1. Do I have to buy a USBtoSerial adapter? AFAIK, Sipeed's MaixPy has WiFi support and they doesn't seem to require an adapter. It would be great if you can explain why it is required.
2. I'm getting an error with the following configuration on macOS Mojave. I guess it is related to the fact that I ignore the whole "**Flashing AT Firmware to the Sipeed-M1 (DanDock) board's WiFi (ESP8285)**" part due to the lack of a USBtoSerial adapter? (You can see there are some bash errors at the top of the error message, though, probably due to the incompatibility between BSD and Linux)
```
SERIAL_PORT="/dev/cu.wchusbserial1430"
FLASH_TYPE="1MB"
FLASH_MODE="dout"
EXOPT="-b 460800"
ERASE_FLASH="yes"
ONLY_BOOT="no"
ONLY_FW="no"
PART_NUM=0
TEST_ONLY="no"
ERASE_ONLY="no"
ONLY_CRT="no"
FLASH_BLANKS=""
FW_FILE=""
```
Error message
```
$ ./flash.sh -t 1MB -m dout
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
./flash.sh: line 217: [: -ge: unary operator expected

==========================
Flashing ESP8266 firmware:
--------------------------
  Flash type: 1MB
  Flash mode: dout
   Partition: 0
     Address: 0x001000
    Firmware: '../upgrade/esp8285_AT_1_2.bin'
 Erase Flash: 'yes'
================================================

esptool.py v2.7
Serial port /dev/cu.wchusbserial1430
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

Erase flash failed!
```
3. Maix M1w seems to have a 8MB flash (?), but current the biggest supported flash size is 4MB. How can I get around with this?
4. Forgive a total newbie like me, but can you explain what this repo is for? I suppose it's to provide WiFi support for MaixPy-FreeRTOS, but you also mentioned "OTA", does it mean I can flash my board over WiFi without connecting it to a computer?

Anyway, this repo is at least as cool as Maixpy. Good job!

pak-man, 2019-08-04:
1- Yes, you need a USB-Serial adapter. Because the ESP8285 modules has its own microcode, and the chip´s Rx/Tx lines are hooked to the Sipeed chip, you need to access those lines to flash the ESP chip. 
2- I don´t know about the specificities of MacOS, but serial flashing the ESP8285 did not work for me too, although I did not have the same error. For me the -m DOUT option was not supported _probably due to the specific version of the ESP8285 mounted on my board (Maix Dock Suit with LCD and camera). Removing the -m DOUT from the flash commandline worked.

3 - don´t know
4- I don´t know about the OTA firmare, but I suppose it allows to flash the ESP8285 firmware only.

nalzok, 2019-08-04:
Try changing `-m DOUT` to `-m dout`?