loboris/MicroPython_K210_LoBo #9: Cannot flash the WiFi firmware to M1W

nalzok, 2019-08-05:
So I have got myself a [USB to Serial Adapter](https://www.amazon.com/gp/product/B00LZV1G6K/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) on Amazon, but I'm still getting the following error. Any tips?

```
$ ./flash.sh -eo -t 1MB -m dout
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
./flash.sh: line 217: [: -ge: unary operator expected

==========================
Flashing ESP8266 firmware:
--------------------------
  Flash type: 1MB
  Flash mode: dout
   Partition: 0
     Address: 0x001000
    Firmware: '../upgrade/esp8285_AT_1_2.bin'
 Erase Flash: 'yes'
================================================

esptool.py v2.7
Serial port /dev/cu.wchusbserial1410
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

Erase flash failed!
```

Here is my set up (quite crude, I know)

[image]
[image]
[image]

pak-man, 2019-08-05:
- It´s not written on the README.md, but you should connect the Ground of your USB-TTL adapter to the Ground of your board as well. Otherwise you end up with floating potential and a non-working serial communication. 
- Also I strongly advice you to solder a magnet wire with a switch to ground to the BOOT pin of the ESP8285 (that one near the antenna): It´s too easy to mess up things and create shortcuts, the way you do. 
- Then you have a look again at the [wiring picture](https://github.com/loboris/MicroPython_K210_LoBo/blob/master/firmware/wifi/flash/Sipeed_DanDock_wifi.jpg) : You´re connecting Rx/Tx to the pins 4 & 5, it should be 6 & 7

nalzok, 2019-08-06:
+ Somehow, it still doesn't work when the Ground is connected, with the same error message. Is there any way to get a more verbose debug output? For example, does a flashing LED on the USB-TTL  adapter mean anything?
+ I don't have a solder at hand, and I'll try to get one, but that'll take some time.

pak-man, 2019-08-06:
Spare also _your_ time and solder proper headers to that board, then use cables that make a **reliable** electrical contact. 
Believe me, here is no chance you will flash a whooping 1.5MB at 57KB/s while holding those pins by hand. 

You´re living in strange world where access to cutting-edge micro-controller is possible, but a simple soldering iron is out of reach. Like owning Formula 1 car but having no asphalt around and wondering why you can´t steer it in the mud.

nalzok, 2019-08-06:
LOL, nice metaphor! Now, storytime if you don't mind...

I'm a statistics major from China working as a Research Assistant in the U.S., so it's probably unsurprising that I don't have a solder in my apartment. The thing is, I'm also an amateur in EECS, so I brought a Maix M1W board from China to have some fun during my stay in the U.S. I thought playing with it just requires a computer and a USB cable, but it clearly turns out to be wrong and I need to buy adapters and solders. While It's not hard to order a solder on Amazon, my roommates would complain about the smell and the risk of burning everything, so I have to find a safer place to do the job, and that takes time.

By the way, I see you have another repo called [MicroPython_ESP32_psRAM_LoBo]( https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo), and there happens to be a pre-soldered [Maixduino](http://dl.sipeed.com/MAIX/HDK/Maixduino/Specifications/) with an ESP32 module at my hand (you may find its name confusing because Sipeed uses "Maixduino" as the name for both their [hardware](http://dl.sipeed.com/MAIX/HDK/Maixduino/) and [software](https://github.com/sipeed/Maixduino)). Is there any chance that I can get them to work together when waiting for the solder?

pak-man, 2019-08-06:
I´m pretty sure you can find a hackerlab around you where you can do that "dirty" job of soldering few pins. The repository for Micropython ESP32 won´t work on the Maixduino. Also it´s pretty outdated, last commit was 10 months ago.
So, 10 minutes soldering with open window is really what you should do. Then close this non-issue.

loboris, 2019-08-06:
@nalzok
You could probably succeed to flash the firmware if you bend the pins from the adapter a little so that you get the good and stable contacts with the board. The ESP8285 `boot` pin has to be connected to the ground only while powering the board up.
As the _amateur in EECS_ you should know that the two (UART) circuits cannot communicate without common ground.
You can test the board even without flashing the WiFi firmware. You just won't be able to use the network functions.

ryanburgett, 2019-12-19:
If you still haven't gotten this to work, or if anyone else has this issue, try switching TX and RX. Sometimes these cheap USB Serial devices are mislabeled.

lypanov, 2020-05-06:
> @nalzok
> You could probably succeed to flash the firmware if you bend the pins from the adapter a little so that you get the good and stable contacts with the board. The ESP8285 `boot` pin has to be [...]

Thank you. Thanks to this explanation (of the need for common ground) I was able to hack at least get the basic soldering done that was needed to hook my K210 up to old 8266 I had laying around. I just wish I'd thought of this 6 months ago! I'll give upgrading the internal 8285 a try eventually when I'm a bit more confident but I have zero background / aptitude in anything related to the physical realm..Happy to be back in software land now with the K210!

nalzok, 2020-10-08:
I have successfully flashed the firmware to the internal ESP8285 after soldering the pins. It took me tens of tries before finally succeeded, so you probably need to re-run the flash script a few times. In fact, I didn't follow the instruction in README thoroughly. The commands given are

```
./flash.sh -eo -t 1MB -m dout -P /dev/cu.usbserial-1430
./flash.sh -t 1MB -m dout -P /dev/cu.usbserial-1430
```

The first line will erase the flash, and the second one will erase the flash (yes, erase again) and then flash the firmware. It appears that you cannot erase/flash the device again without doing a power cycle, so I used the following commands instead

```
./flash.sh -eo -t 1MB -m dout -P /dev/cu.usbserial-1430
./flash.sh -ne -t 1MB -m dout -P /dev/cu.usbserial-1430
```

The second line now has an additional option `-ne`, which means "No erase", so it will do the flashing without erasing it again first.

Note that by close visual inspection, the pins touch the pad firmly, so the flashing errors like `A fatal error occurred: Timed out waiting for packet header` are probably not due to poor contact.

[image]