loboris/MicroPython_K210_LoBo #10: GPIO reset

devrosx, 2019-08-09:
hello tried this simple code conected with pullup button (on M1W)
`from machine import Pin
p1 = Pin(18, Pin.IN, Pin.PULL_UP)
while True:
    if not p1.value():
        print('Button pressed!')

print(p1.value())`
when i pres button Maixpy is reseted...
tried same code on esp32 and works

loboris, 2019-08-12:
Sorry for late reply.

The reset has nothing to do with the Pin state.
It occurs because the the watchdog resets the system after 6 seconds.

Every **long running loop** should have either the [sleep_ms](https://github.com/loboris/MicroPython_K210_LoBo/wiki/utime#utimesleep_ms-ms-) method with period longer tham 2 ms, or [reset watchdog method](https://github.com/loboris/MicroPython_K210_LoBo/wiki/machine#machinewdt_reset) inserted (or both).

```python
import time, machine
from machine import Pin
p1 = Pin(18, Pin.IN, Pin.PULL_UP)
while True:
    if not p1.value():
        print('Button pressed!')
        while not p1.value():
            #time.sleep_ms(3)
            machine.wdt_reset()
        print('Button not pressed!')
    #time.sleep_ms(3)
    machine.wdt_reset()
```
```console
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time, machine
=== from machine import Pin
=== p1 = Pin(18, Pin.IN, Pin.PULL_UP)
=== while True:
===     if not p1.value():
===         print('Button pressed!')
===         while not p1.value():
===             #time.sleep_ms(3)
===             machine.wdt_reset()
===         print('Button not pressed!')
===     #time.sleep_ms(3)
===     machine.wdt_reset()
=== 
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Button pressed!
Button not pressed!
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
KeyboardInterrupt: 
>>> 
```

devrosx, 2019-08-29:
thanks problem solved :)

closed 2019-08-29 by devrosx