loboris/MicroPython_K210_LoBo #11: KPU

andriyadi, 2019-08-24:
This is definitely a better MPy port for K210 for Pythonic programming. I can explore a lot of MPy modules and features. 
But regarding to access the KPU, I'm not sure how to do it. Please advice.

loboris, 2019-08-26:
I have just finished the `Camera` module.
**KPU** support will be added soon, probably during the next month.

CorruptName, 2019-12-24:
> I have just finished the `Camera` module.
> **KPU** support will be added soon, probably during the next month.

Was this ever added? If so, where can I find more info on how to use it?

loboris, 2019-12-24:
It will be incleded in the next update, in the next couple of days.
The documentation will be available on the Wiki pages.

krishnak, 2020-01-08:
Looking forward to it, the star attraction

kartben, 2020-01-24:
Hi @loboris, have you been able to make progress on KPU support? Anything we can do to help? Cheers!

loboris, 2020-01-28:
@kartben I'm just preparing the update, it will be commited today or tomorrow.
It include the full cammera support, but I've postponed the KPU module (I don't use it at the moment, and I have too many other tasks to do...), I plan to work on it in a month or two, sorry.

NeuerUser, 2020-12-22:
Still no KPU support (😢). I could understand if it were lots of work. But it's very strange that it was promised so often over such a long time and was always 'ready to be committed" or "just preparing the update".

Without a KPU the whole pet does not make mich sense for me. The KPU is the major differentiator for the K210.

loboris, 2020-12-22:
@NeuerUser

> Without a KPU the whole pet does not make mich sense for me. The KPU is the major differentiator for the K210.

I can't agree with that.
K210 is an amazing microcontroller even if used as "general purpose" one.
I'm not using KPU in any of my projects at the moment, so it is of low priority for me.

Anyway, the KPU support will be probably added in January, but I'll not promise this time...
Other updates (many of them, including ESP32 support) will be commited in a couple of days.