loboris/MicroPython_K210_LoBo #12: Using pre-built firmware with Maixduino

baqwas, 2019-09-03:
Hello,
I'm having teething problems deploying the _pre-built firmware_ to the [**Maixduino**](https://maixduino.sipeed.com/en/) board. The following command from the _unarchived_ firmware working directory reports that **K210** cannot be detected!
`./kflash.py -p /dev/ttyUSB0 -b 2000000 -t MaixPy.bin`

I have attached a document listing the steps performed to arrive at the last command. I would appreciate some guidance on troubleshooting the issue. In the interim, I am trying to build the firmware for a subsequent test as a workaround activity.

[MicroPython - installing pre-built firmware.docx](https://github.com/loboris/MicroPython_K210_LoBo/files/3569787/MicroPython.-.installing.pre-built.firmware.docx)

loboris, 2019-09-04:
Please see [this](https://github.com/loboris/MicroPython_K210_LoBo/issues/13#issuecomment-527757514).

baqwas, 2019-09-05:
Problem per suggestion by @loboris on another thread to use a lower baud rate owing to known issue with USB/serial interface. Used 115200 and it worked fine but had to RST the board before continuing use in REPL mode.

closed 2019-09-05 by baqwas