loboris/MicroPython_K210_LoBo #13: Local build firmware

baqwas, 2019-09-03:
Hello,

This issue _superficially_ may be different and therefore with apologies I have started a new thread. I cannot flash a local build firmware to the **Maixduino** board. The flash process cannot detect K210 which is similar to that I reported using the pre-built firmware.

This time around I changed cables and ports and verified that irrespective of the cable and the physical port, the attached device (_viz._ Pixel 2XL) is detected and participates in bi-directional data exchange.
[MicroPython local build firmware.docx](https://github.com/loboris/MicroPython_K210_LoBo/files/3570790/MicroPython.local.build.firmware.docx)

So I did some nominal dump of the port information that has been captured in the attachment. Bottom line, I suspect that the **Sipeed** implementation of the FTDI driver leaves something to be improved later. Is my _amateur_ interpretation correct? Appreciate your indulgence and further guidance.

Kind regards.

lewisxhe, 2019-09-03:
I think you should try with a lower baud rate, because Maixduino uses CH552, he does not support baud rates higher than 1.500000.

baqwas, 2019-09-04:
@lewisxhe 
Thanks for the suggestion. I parametrically ran down from 1,000,000 to 115,200 (presumably the default) without any success.

```
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ ./kflash.py -p /dev/ttyUSB0 -b 1000000 -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
...............
[ERROR] No vaild Kendryte K210 found in Auto Detect, Check Your Connection or Specify One by`-p /dev/ttyUSB0` 
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ ./kflash.py -p /dev/ttyUSB0 -b 115200 -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
...............
[ERROR] No vaild Kendryte K210 found in Auto Detect, Check Your Connection or Specify One by`-p /dev/ttyUSB0` 
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ ./kflash.py -p /dev/ttyUSB0 -b 750000 -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
...............
[ERROR] No vaild Kendryte K210 found in Auto Detect, Check Your Connection or Specify One by`-p /dev/ttyUSB0` 
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ 

```

Do you think that Sipeed's implementation of the FTDI (if there is a such a thing) is causing an issue under Ubuntu 19.04? I don't have access to any other platform except Raspbian but I would have to procure a cable for that test (I presume). Currently, I have a TypeC-TypeC and a TypeA-TypeC set of cables. Perhaps I could build the firmware under Raspbian and use this second cable to the Maixduino. Honestly, these things are out of my league.

Kind regards.

lewisxhe, 2019-09-04:
Ubuntu I don't know, I am using the windows wls liunx subsystem, or you can try to use [kflash_gui]https://github.com/sipeed/kflash_gui/releases for burning
@baqwas

baqwas, 2019-09-04:
@lewisxhe 

Thanks a bunch for the link. I will definitely review the site and choose one that is comprehensible for my limited understanding. Appreciate your suggestion very much. If I am successful, I will definitely let you know.

Kind regards.

loboris, 2019-09-04:
It doesn't really matter (for your issue) which baudrate you select. K210 is always detected at 115200 and the bootloader firmware is first uploaded at that speed. Only then the firmware is uploaded at the selected baud rate.
The upload process starts with activating K210 boot mode. Two K210 pins are used to enter the bootloader mode. `BOOT` pin must be held low and `RST` pin must be pulled low then released.
Those pins are connected to the CH552 (USB to UART chip on Maixduino board ([schematic](http://dl.sipeed.com/MAIX/HDK/Maixduino/Maixduino-4.30/Maixduino-4.30%28schematic%29.pdf))) pins `1` and `2` and are activated by the `kflash`.
The pins are also connected to the **buttons** `BOOT` and `RST` so you can enter the bootloader mode manually:
When you see the `[INFO] Trying to Enter the ISP Mode...` message press the `BOOT` button, then press the `RST` button and release it, release `BOOT` button.
If the upload starts, than something is probably wrong with the CH552...

You can also try to upload the firmware using the **`-B goE`** argument, it may be needed for CH552:
```
./kflash.py -p /dev/ttyUSB0 -b 1000000 -B goE -t MaixPy.bin
```

baqwas, 2019-09-04:
Hello @loboris 

Your observation about the starting baud rate is spot on because of the prompts in the terminal window. ;)

Unfortunately, `-B goE` option did not resolve the issue as the following output shows:
```
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ ./kflash.py -p /dev/ttyUSB0 -b 1000000 -B goE -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
*
[INFO] Greeting Message Detected, Start Downloading ISP 
Downloading ISP: |=============================================| 100.0% 10kiB/s
[INFO] Booting From 0x80000000 
[INFO] Wait For 0.1 second for ISP to Boot 
[INFO] Booted to Flashmode Successfully 
[INFO] Selected Baudrate:  1000000 
[INFO] Baudrate changed, greeting with ISP again ...  
[WARN] Timeout Error, retrying... 
...
[WARN] Timeout Error, retrying... 
[ERROR] Failed to Connect to K210's Stub 
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$
```

Subsequently, I tried your other suggestion - manual dexterity with small and my fat fingers may not complement each other but just to be sure I repeated the exercise three times and the output was almost identical - exception raised the program that is running.
```
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ ./kflash.py -p /dev/ttyUSB0 -b 1000000 -B goE -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
*
[INFO] Greeting Message Detected, Start Downloading ISP 
Traceback (most recent call last):=======----------------------| 52.3% 10kiB/s
  File "./kflash.py", line 1358, in <module>
    main()
  File "./kflash.py", line 1355, in main
    kflash.process()
  File "./kflash.py", line 1257, in process
    loader.install_flash_bootloader(isp_loader)
  File "./kflash.py", line 969, in install_flash_bootloader
    self.flash_dataframe(data, address=0x80000000)
  File "./kflash.py", line 898, in flash_dataframe
    if self.recv_debug():
  File "./kflash.py", line 806, in recv_debug
    op, reason, text = ISPResponse.parse(self.recv_one_return())
  File "./kflash.py", line 606, in recv_one_return
    raise TimeoutError
__main__.TimeoutError
reza@BeUlta:~/projects/MicroPython_K210_LoBo/k210-freertos$ 

```

I will gladly execute other commands upon your suggestion. Thanks a bunch again.

Kind regards.

loboris, 2019-09-04:
**The ISP (boot loader) is now started!**
I've found that the CH552 has problems with higher baud rates ([look here](https://github.com/diodep/ch55x_dualserial)), so you may try  with baudrates set to 750000, 500000, 250000 and 115200 (**`-b`** argument), as suggested.
I think at some speed the upload will be successfull.

baqwas, 2019-09-04:
Bingo! The -B option with the lower rate works!!! THANK YOU VERY MUCH.
[image]
I started with 115200 and it worked. There is no need for me to go higher right now.

Can you please point me in the right direction to work with MicroPython? **I won't bug you** but I need a jump start (e.g. which IDE/terminal window, essential libraries, a sample or two). I am conversant with Python ;) which is why I jumped at your offer of assistance to deploy MicroPython.

Kind regards.

closed 2019-09-04 by baqwas

loboris, 2019-09-04:
I'm glad you have finally succeded in flashing the firmware.

It looks like your board was not **reset** after firmware upload. You should get something like this:
```console
M (1371) [MAIXPY]: Configuration loaded from flash
M (9298) [MAIXPY]: Heaps: FreeRTOS=3840 KB (1196 KB free), MPy=2560 KB, other=379 KB

 __  __              _____  __   __  _____   __     __ 
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ / 
| |\/| |   / /\ \     | |     > <   |  ___/    \   /  
| |  | |  / ____ \   _| |_   / . \  | |         | |   
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.4
-------------------------------
MicroPython 1.11.4 (ac4b72e-dirty) built on 2019-07-09; Sipeed_board with Kendryte-K210
Type "help()" for more information.
```
You may try to press the **reset button** to reset it manually.
After the reset you are in the MicroPython `REPL` (Interactive Interpreter Mode) which is very much simmilar to any other Python.
For start you can consult the [MicroPython Documentation](https://docs.micropython.org/en/latest/index.html) and the [Wiki pages](https://github.com/loboris/MicroPython_K210_LoBo/wiki) from this repository (not complete yet, but I'm working on it). Both have some code examples you can try.

Terminal emulator is provided if you cloned the repository or downloaded the firmwares archive.
Just execute:
```console
./MPyTerm.py -d /dev/ttyUSB0 -r True
```
Again, if the board is not reset, reset it manually.

baqwas, 2019-09-05:
Yes, my enthusiasm dried up after I sent the previous note. I did not try RST alone but in my _infinite wisdom_ I did the BOOT/RST sequence not knowing what to expect and saw nothing. Then I read your note and pressed only the RST button. Then, _voila_!
`MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (1abe503-dirty) built on 2019-09-03; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>> 
`
I now have the familiar Python prompt and I have briefly reviewed the available libraries (and noted the _u_ prefix for some of these). I can keep myself busy for a few hours a week.

BTW, I noticed that you could with an extra pair of hands for documentation. If you have some simple things (e.g. basic test exercises), please do not hesitate to suggest something for me to focus on. I will gladly try to contribute.

Once again, I cannot thank you enough for your patience in helping out a newbie like myself.

Kind regards.