loboris/MicroPython_K210_LoBo #14: FLASH File system initialization failed!

ryanburgett, 2019-12-17:
I have tried flashing 3 times and i keep getting this issue:
Board: Sipeed MAiX Dock M1W
Trying to get extra features such as ftp and ota which are not supported in official fw.

`$ kflash -p /dev/ttyS7 -b 2000000 -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyS7
[INFO] Default baudrate is 115200 , later it may be changed to the value you set.
[INFO] Trying to Enter the ISP Mode...
._._._._.
[INFO] Automatically detected dan/bit/trainer

[INFO] Greeting Message Detected, Start Downloading ISP
Downloading ISP: |=====================================================================================| 100.0% 10kiB/s
[INFO] Booting From 0x80000000
[INFO] Wait For 0.1 second for ISP to Boot
[INFO] Boot to Flashmode Successfully
[INFO] Selected Baudrate:  2000000
[INFO] Baudrate changed, greeting with ISP again ...
[INFO] Boot to Flashmode Successfully
[INFO] Selected Flash:  On-Board
[INFO] Initialization flash Successfully
Programming BIN: |=====================================================================================| 100.0% 60kiB/s
[INFO] Rebooting...
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyS7  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

W (1245) CONFIG: Error reading configuration (crc)
M (22232) CONFIG: Default flash configuration set (1)
Q (28226) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (35868) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595c80
M (44238) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB
lfs error:969: Corrupted dir pair at 0 1
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
E (10078601) [MAIN]: FLASH File system initialization failed!`

Any Advice?

ryanburgett, 2019-12-18:
I even built firmware from source and still no-go. Should i try with SD card?

ryanburgett, 2019-12-18:
This is what it says the first time. Also tried flashinging "erase_spiffs.kfpkg" first. No help.
```
MPY: soft reboot
lfs error:969: Corrupted dir pair at 1 0
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
Erasing Flash, this can take some time...
Flash erased
lfs debug:1604: Bad block at 0
lfs warn:1609: Superblock 1 has become unwritable
Error formating (-28)
E (38698954) [MAIN]: FLASH File system initialization failed!
```

loboris, 2019-12-19:
I can't reproduce this issue on Sipeed MAiX Dock M1W.
Have you made any changes to the sources?
Have you tried to flash the prebuilt firmware?

**It looks you are not using the `kflash.py` from this repository, try to use it.**

My test with prebuilt firmware:
- Download and unpack _MaixPy_firmware.zip_.
- erase Sipeed MAiX Dock M1W flash (to be shure the file system is not formated)
```
boris@UbuntuMate:/home/LoBo2_Razno/K210_razvoj/xxxxx$ ./kflash.py -E
[INFO] COM Port Auto Detected, Selected  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
...
...
[INFO] Initialized flash Successfully 
[INFO] Erasing SPI Flash... 
[INFO] SPI Flash erased. 
```
- Flash the firmware
```
boris@UbuntuMate:/home/LoBo2_Razno/K210_razvoj/xxxxx$ ./kflash.py -p /dev/ttyUSB0 -b 2000000 -t MaixPy.bin
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
...
...
[INFO] Flashing data at Flash address 0x00000000 
[INFO] Flashing with SHA prefix 
Programming BIN:
...
[INFO] Rebooting... 

--[ MicroPython terminal  ver. 5.1.3 ]-- 
--[ Press ESC twice for command mode ]-- 

W (1243) CONFIG: Error reading configuration (crc)
M (22232) CONFIG: Default flash configuration set (1)
Q (28225) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (35867) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595cc0
M (44237) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB
lfs error:969: Corrupted dir pair at 0 1
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
Littlefs formated and mounted

 __  __              _____  __   __  _____   __     __ 
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ / 
| |\/| |   / /\ \     | |     > <   |  ___/    \   /  
| |  | |  / ____ \   _| |_   / . \  | |         | |   
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (76c3fd1-dirty) built on 2019-08-07; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>> 
```
(pressed **F** on fs error prompt.

---

How exactly are you "trying to get extra features such as ftp and ota" ?

The repository wasn't updated for some time, but I have a large amount of changes waiting to be commited, I'll do it before Christmas.
Among other changes, FTP and OTA will be available in the new release.

ryanburgett, 2019-12-19:
Hello! Thank you for your response and information. Thanks for all you do and your contributions to the community! Much appriciated, even if I'm not able to get this fw working. Sorry, I thought those changes had already been committed. No worries. Still, your fork is way ahead of official FW (directories in SPIFFS, lol.) and of course I started with the pre-compiled FW! I'm way to lazy to start with compiling.

I did exactly as you instructed (MANY MANY TIMES), both from WSL and actual Debian. The results are the same. Thanks again, but still no go. Note: It seems to "do more" when i choose "E" from the menu.

```
~ ⌚ 6:13:52
$ mkdir tmp

~ ⌚ 6:13:55
$ cd tmp

~/tmp ⌚ 6:14:00
$ wget https://github.com/loboris/MicroPython_K210_LoBo/raw/master/firmware/MaixPy_firmware.zip
--2019-12-19 06:14:07--  https://github.com/loboris/MicroPython_K210_LoBo/raw/master/firmware/MaixPy_firmware.zip
Resolving github.com (github.com)... 140.82.113.3
Connecting to github.com (github.com)|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/loboris/MicroPython_K210_LoBo/master/firmware/MaixPy_firmware.zip [following]
--2019-12-19 06:14:08--  https://raw.githubusercontent.com/loboris/MicroPython_K210_LoBo/master/firmware/MaixPy_firmware.zip
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.200.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.200.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8866200 (8.5M) [application/zip]
Saving to: ‘MaixPy_firmware.zip’

MaixPy_firmware.zip           100%[=================================================>]   8.46M  7.15MB/s    in 1.2s     
2019-12-19 06:14:09 (7.15 MB/s) - ‘MaixPy_firmware.zip’ saved [8866200/8866200]

~/tmp ⌚ 6:14:10
$ unzip MaixPy_firmware.zip
Archive:  MaixPy_firmware.zip
   creating: wifi/
   creating: wifi/flash/
  inflating: wifi/flash/Sipeed_DanDock_wifi.jpg
  inflating: wifi/flash/flash.sh
  inflating: wifi/flash/README.md
   creating: wifi/upgrade/
  inflating: wifi/upgrade/version.txt
  inflating: wifi/upgrade/esp8285_AT_1_2.bin
  inflating: wifi/upgrade/esp8285_AT_2_2.bin
 extracting: wifi/upgrade/esp8285_AT_1_2.md5
 extracting: wifi/upgrade/esp8285_AT_2_2.md5
  inflating: wifi/upgrade/esp8266_AT_1_0.bin
 extracting: wifi/upgrade/esp8266_AT_1_0.md5
  inflating: wifi/upgrade/esp8266_AT_1_2.bin
 extracting: wifi/upgrade/esp8266_AT_1_2.md5
  inflating: wifi/upgrade/esp8266_AT_2_2.bin
 extracting: wifi/upgrade/esp8266_AT_2_2.md5
  inflating: wifi/upgrade/esp8266_AT_1_3.bin
 extracting: wifi/upgrade/esp8266_AT_1_3.md5
  inflating: wifi/upgrade/esp8266_AT_2_3.bin
 extracting: wifi/upgrade/esp8266_AT_2_3.md5
  inflating: wifi/upgrade/esp8266_AT_1_4.bin
 extracting: wifi/upgrade/esp8266_AT_1_4.md5
  inflating: wifi/upgrade/esp8266_AT_2_4.bin
 extracting: wifi/upgrade/esp8266_AT_2_4.md5
  inflating: wifi/upgrade/esp8266_AT_1_5.bin
 extracting: wifi/upgrade/esp8266_AT_1_5.md5
  inflating: wifi/upgrade/esp8266_AT_1_6.bin
 extracting: wifi/upgrade/esp8266_AT_1_6.md5
  inflating: wifi/blank.bin
  inflating: wifi/bootloader_noota.bin
 extracting: wifi/bootloader_noota.md5
  inflating: wifi/bootloader.bin
 extracting: wifi/bootloader.md5
 extracting: wifi/bootloaderver.txt
  inflating: wifi/esp_ca_cert.bin
  inflating: wifi/esp_init_data_default_v08.bin
  inflating: wifi/README.md
  inflating: kflash.py
[... 161 lines cut ...]
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (76c3fd1-dirty) built on 2019-08-07; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>>
MPY: soft reboot
lfs error:969: Corrupted dir pair at 0 1
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
lfs debug:1604: Bad block at 0
lfs warn:1609: Superblock 1 has become unwritable
Error formating (-28)
E (12947238) [MAIN]: FLASH File system initialization failed!
```
(Pressed F)

I will add that through testing I did get this working once - BUT - it would fail to mount every other boot and had no SD Support.

Also, i have only had this board 2 days now. Is it possible they changed something? "The official" fw seems to work fine. Is it possible that it is as simple as needing a delay for the flash to init? Seems like that is possible bc of the intermittent behavior (when the flash does init).

Thanks again! Hopefully you have something else i can try? (BTW, I first did all these steps on Tinkerboard running Debian. - same results.)

loboris, 2019-12-19:
This is strange, I've tested it on 4 different K210 board and never had a similar issue.
This port uses much higher Flash clock, it may be the reason for the issue (it may be hardware related).
I'll try to check the issue more...

> Try to flash the **MaixPy.kfpkg** from `MaixPy_firmware.zip` instead of only the **MaixPy.bin**. It contains the precompiled file system which will be flashed to the right position so the firmware should detect it at start.

```
boris@UbuntuMate:/home/mpy_test$ ./kflash.py -p /dev/ttyUSB0 -b 2000000 -t MaixPy.kfpkg
[INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO] Trying to Enter the ISP Mode... 
.
[INFO] Greeting Message Detected, Start Downloading ISP 
[INFO] CH340 mode 
Downloading ISP: |===============================================| 100.0% 10kiB/s
[INFO] Booting From 0x80000000 
[INFO] Wait For 0.1 second for ISP to Boot 
[INFO] Booted to Flashmode Successfully 
[INFO] Selected Baudrate:  2000000 
[INFO] Baudrate changed, greeting with ISP again ...  
[INFO] Booted to Flashmode Successfully 
[INFO] Selected Flash:  On-Board 
[INFO] Initialized flash Successfully 
[INFO] Extracting KFPKG ...  
[INFO]   Writing MaixPy.bin to Flash address 0x00000000 
[INFO] Flashing data at Flash address 0x00000000 
[INFO] Flashing with SHA prefix 
Programming BIN: |===============================================| 100.0% 65kiB/s
[INFO]   Writing maixpy_lfs.img to Flash address 0x00400000 
[INFO] Flashing data at Flash address 0x00400000 
Programming BIN: |===============================================| 100.0% 75kiB/s
[INFO] Rebooting... 

--[ MicroPython terminal  ver. 5.1.3 ]-- 
--[ Press ESC twice for command mode ]-- 

M (3216) [MAIXPY]: Configuration loaded from flash
Q (7732) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (16065) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595cc0
M (25229) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB

 __  __              _____  __   __  _____   __     __ 
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ / 
| |\/| |   / /\ \     | |     > <   |  ___/    \   /  
| |  | |  / ____ \   _| |_   / . \  | |         | |   
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (76c3fd1-dirty) built on 2019-08-07; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>> 
>>> import os
>>> os.listdir('flash')
['boot.py', 'examples', 'fonts', 'lib', 'www']
>>> 
```

ryanburgett, 2019-12-19:
I think i have narrowed down the issue a bit. It seems almost certainly timing related.. If I repeatedly push the reset button, 3/10 times, the flash will initialize.

Erase:
```
[INFO] COM Port Selected Manually:  /dev/ttyS7
[INFO] Default baudrate is 115200 , later it may be changed to the value you set.
[INFO] Trying to Enter the ISP Mode...
.......
[INFO] Greeting Message Detected, Start Downloading ISP
Downloading ISP: |=====================================================================================| 100.0% 10kiB/s
[INFO] Booting From 0x80000000
[INFO] Wait For 0.1 second for ISP to Boot
[INFO] Booted to Flashmode Successfully
[INFO] Selected Flash:  On-Board
[INFO] Initialized flash Successfully
[INFO] Erasing SPI Flash...
[INFO] SPI Flash erased.
```
Flash:
```
~/tmp ⌚ 7:48:55
$ ./kflash.py -p /dev/ttyS7 -b 2000000 -t MaixPy.kfpkg
[INFO] COM Port Selected Manually:  /dev/ttyS7
[INFO] Default baudrate is 115200 , later it may be changed to the value you set.
[INFO] Trying to Enter the ISP Mode...
......
[INFO] Greeting Message Detected, Start Downloading ISP
[INFO] CH340 mode
Downloading ISP: |=====================================================================================| 100.0% 10kiB/s
[INFO] Booting From 0x80000000
[INFO] Wait For 0.1 second for ISP to Boot
[INFO] Booted to Flashmode Successfully
[INFO] Selected Baudrate:  2000000
[INFO] Baudrate changed, greeting with ISP again ...
[INFO] Booted to Flashmode Successfully
[INFO] Selected Flash:  On-Board
[INFO] Initialized flash Successfully
[INFO] Extracting KFPKG ...
[INFO]   Writing MaixPy.bin to Flash address 0x00000000
[INFO] Flashing data at Flash address 0x00000000
[INFO] Flashing with SHA prefix
Programming BIN: |=====================================================================================| 100.0% 60kiB/s
[INFO]   Writing maixpy_lfs.img to Flash address 0x00400000
[INFO] Flashing data at Flash address 0x00400000
Programming BIN: |=====================================================================================| 100.0% 59kiB/s
[INFO] Rebooting...

--[ MicroPython terminal  ver. 5.1.3 ]--
--[ Press ESC twice for command mode ]--

W (1241) CONFIG: Error reading configuration (crc)
M (22231) CONFIG: Default flash configuration set (1)
Q (28224) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (35866) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595cc0
M (44235) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB
lfs error:969: Corrupted dir pair at 0 1
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
Erasing Flash, this can take some time...
Flash erased
lfs debug:1604: Bad block at 0
lfs warn:1609: Superblock 1 has become unwritable
Error formating (-28)
E (27433582) [MAIN]: FLASH File system initialization failed!

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
[... 460 lines cut ...]
Fail.
Hard Reset (Button)
```
W (1244) CONFIG: Error reading configuration (crc)
M (22231) CONFIG: Default flash configuration set (1)
Q (28225) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (35867) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595cc0
M (44236) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (76c3fd1-dirty) built on 2019-08-07; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>>
```
Success!
As you can see, 50/50 shot as to if it mounts or not. I don't think we have SD support either, need to test.

ryanburgett, 2019-12-19:
After more testing, There are more issues but probably still related to speed/timing.
```python
>>> os.listdir()
['boot.py', 'examples', 'fonts', 'lib', 'www']
>>> os.listdir()
['boot.py', 'examples', 'fonts', 'lib', 'www']
>>> os.listdir()
['boot.py', 'examples', 'fonts', 'lib', 'www']
>>> os.listdir()
lfs error:969: Corrupted dir pair at 0 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -1
>>> os.listdir()
lfs error:969: Corrupted dir pair at 0 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -1
>>> os.listdir()
lfs error:969: Corrupted dir pair at 1 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -1
>>> os.listdir()
['boot.py', 'examples', 'fonts', 'lib', 'www']
```

ryanburgett, 2019-12-19:
Do you agree that they probably used a slower flash on the board that i have and that means that it cannot keep up with the fw which has been optimized for faster flash?

loboris, 2019-12-19:
Yes, I think it is a hardware issue.

You can try to change the default SPI Flash clock in **k210-freertos/platform/drivers/include/w25qxx.h**:
`#define WQ25QXX_MAX_SPEED                   83000000`
Change to 40000000 or 25000000 and rebuild.

ryanburgett, 2019-12-19:
Works perfectly now @ 40000000. Except No SD Card.
Do i need to mount SD manually?

ryanburgett, 2019-12-19:
```javascript
M (3218) [MAIXPY]: Configuration loaded from flash
Q (7735) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (16067) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595c80
M (25232) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB
lfs error:969: Corrupted dir pair at 0 1
Error mounting Flash file system, select an option
A - continue without file system (default)
F - format Flash and try to mount again
E - erase Flash, format and try to mount again
Littlefs formated and mounted

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (1abe503-dirty) built on 2019-12-19; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>>
M (3221) [MAIXPY]: Configuration loaded from flash
Q (7738) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (16070) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595c80
M (25235) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (1abe503-dirty) built on 2019-12-19; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>>
```
Hard Reset (Button):
```
M (3215) [MAIXPY]: Configuration loaded from flash
Q (7731) [MAIXPY]: PLL0=988000000, PLL1=398666666, PLL2=45066666, SPI3clk=494000000
Q (16064) [MAIXPY]: RAM buffer of 1024 bytes allocated at 0x80595c80
M (25229) [MAIXPY]: Heaps: FreeRTOS=3716 KB (811 KB free), MPy=2820 KB, other=419 KB

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (1abe503-dirty) built on 2019-12-19; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>>
```
Persists reboots and:
```python
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
>>> os.listdir()
['flash']
```
Reliably reads fs. YAY!

loboris, 2019-12-19:
I'll include the option to set the maximal SPI Flash clock speed in the next update.

SD Card is not mounted automatically, you have to mount it manually.
If I remember well, it should be something like this:
```python
import os
sdcard = os.VfsSDCard()
os.mount(sdcard, '/sd')
```
SD card must be formated as fat32.

loboris, 2019-12-19:
I've tested SD Card support and it works as expected:

```python
>>> import os
>>> sdcard = os.VfsSDCard()
>>> os.mount(sdcard, '/sd')
>>> os.listdir('/sd')
['boot.py', 'examples', 'fonts', 'lib', 'www']
>>> from upysh import *

upysh is intended to be imported using:
from upysh import *

To see this help text again, type "man".

upysh commands:
pwd, cd("new_dir"), ls, ls(...), head(...), cat(...)
newfile(...), mv("old", "new"), rm(...), mkdir(...), rmdir(...)
cp("from", "to")
clear

>>> ls('/sd')
      82  2019/07/16 15:27:48  boot.py
   <dir>  2019/07/16 10:34:06  examples
   <dir>  2019/07/17 14:36:44  fonts
   <dir>  2019/07/16 10:29:00  lib
   <dir>  2019/07/16 10:29:26  www

Total in '/sd': 82 B
Drive: '/sd' size: 7.387 GB, free: 7.385 GB
>>> ls('/sd/fonts')
    3052  2016/08/13 23:21:08  BigFont.fon
    1166  2016/08/13 23:21:30  DejaVuSans12.fon
    1836  2016/08/14 00:01:12  DejaVuSans18.fon
    2732  2016/08/13 23:22:36  DejaVuSans24.fon
    4192  2016/08/14 00:00:56  DotMatrix_M.fon
   64052  2018/11/10 15:51:28  Georgia128.fon
   13692  2016/08/14 00:02:50  Grotesk24x48.fon
   24057  2016/08/11 14:53:40  OCR_A_Extended_M.c
    3052  2016/08/13 23:20:44  arial_bold.fon
    3459  2019/07/17 14:32:46  dot32.fon
    4008  2019/07/17 14:36:44  lobo30.fon
    3052  2016/08/14 00:03:44  swiss721_outline.fon

Total in '/sd/fonts': 125.34 KB
Drive: '/sd' size: 7.387 GB, free: 7.385 GB
>>> 
```

Note: SD card works in SPI mode. Not all SD cards works good in this mode. I've had the best results with SanDisk 8GB cards.

ryanburgett, 2019-12-19:
Thanks! That was the problem. It wasn't obvious to me that I needed to mount the SD manually since the OFW did this at boot. Great! Glad to have everything working. ALMOST, LOL. I'm having an issue with wifi. Should i start a new issue so that you can mark this one solved? I already flashed the firmware, but it keeps telling me it failed to set baud rate..

ryanburgett, 2019-12-19:
I was thinking, perhaps you should decrease the SPI clock in your next release for compatibility and allow some users to use an option to increase theirs? I bet most people wouldn't care much and this could cause a ton of issues for people, issues which are hard to diagnose. Just my two cents.

closed 2019-12-21 by CorruptName