loboris/MicroPython_K210_LoBo #15: WiFi "Error setting baudrate"

ryanburgett, 2019-12-19:
After successfully flashing the provided firmware, When I try to connect with this error.
```python
>>> import network
>>> wifi = network.wifi
>>> wifi.reset()
>>> wifi.debug(True)
True
>>> wifi.reset()
>>> import _thread, network, time, machine, socket, gc, os
>>> wifi.start(tx=7, rx=6, ssid="SSID", password="PASS", baudrate=921600, wait=True)
M (10359821780) [WIFI]: Initialize WiFi UART
M (10359825780) [WIFI]: UART #0: initialize uart hardware
M (10359856677) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=921641
M (10359882693) [WIFI]: WiFi TASK STARTED
M (10359985755) [ATCMD]: Check and set device baudrate
W (10360626108) [ATCMD]: Trying at 115200 bd
W (10361316499) [ATCMD]: Trying at 230400 bd
W (10362006892) [ATCMD]: Trying at 460800 bd
W (10362697281) [ATCMD]: Trying at 1000000 bd
W (10363387678) [ATCMD]: Trying at 2000000 bd
E (10364076089) [ATCMD]: Error setting baudrate
M (10364080371) [WIFI]: UART #0 deinitialized
M (10364085209) [WIFI]: WiFi TASK TERMINATED
False
```

CorruptName, 2019-12-20:
Okay, So now I'm about to give up..
Got no wifi connection at all.
Tried to flash back to ESP8285's Original Firmware, from:
http://dl.sipeed.com/MAIX/factory_firmware/esp8266/WiFiSPIESP.ino.esp8285_40M_freq_1M_DOUT.bin and loaded Sipeed's Maix 5.0 Firmware so everything should be stock.
Now When i try to connect to wifi, I get ```Stopped Because: couldn't init nic esp8285 ,try again please```
Now i can't get wifi at all. What's wrong here?

loboris, 2019-12-21:
It looks the WiFi firmware flash was OK. It is strange that you can't get any response from ESP8285.
I'll try to look into it. I'm using M1W for 10 months and newer had a problem.

I supose you have disconnected usb->uart adapter before testing.
You can try to init with
```python
wifi.start(tx=7, rx=6, ssid="SSID", password="PASS", baudrate=115200, wait=True)
```

ESP8285 enable pin is connected to K210 io Pin #8 and pulled-up with 12K resistor. You must newer use Pin #8 while using WiFi.
Yo can try to configure the Pin#8 as output and set it high before initializing WiFi (using machine.Pin).

ESP8285 must be initialized even with the original firmware, there should be responses to the commands...

At the moment I can't think of anything else, I'll report if I'll find something.

loboris, 2019-12-21:
Thanks for the comments about flashing process. I'll use them to update the instructions for the next update.

CorruptName, 2019-12-21:
I submitted a pull request for changes to Readme.md if you would like to look at it.
I will try your suggestions.

CorruptName, 2019-12-21:
So this is what i get on the original firmware:
```
>>> fm.register(8, fm.fpioa.GPIOHS0, force=True)
1
>>> wifi_en=GPIO(GPIO.GPIOHS0, GPIO.OUT)
>>>
>>> fm.register(board_info.WIFI_RX, fm.fpioa.UART2_TX, force=True)
1
>>> fm.register(board_info.WIFI_TX, fm.fpioa.UART2_RX, force=True)
1
>>> uart = UART(UART.UART2,115200, timeout=1000, read_buf_len=4096)
>>>
>>> wifi_en.value(1)
>>> nic = network.ESP8285(uart)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: couldn't init nic esp8285 ,try again please
```

loboris, 2019-12-21:
Was it ever working?

CorruptName, 2019-12-21:
> "_I'm just confused why if i set everything back to factory it still does not work?_"
> Was it ever working?

Yes. Definitely.

CorruptName, 2019-12-21:
now i'm back on your firmware, wifi and k210:
```python
 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|
------------------------------------------------------

MaixPy-FreeRTOS by LoBo v1.11.11
--------------------------------
MicroPython 1.11.11 (1abe503-dirty) built on 2019-12-19; Sipeed_board with Kendryte-K210
Type "help()" for more information.
>>> from machine import Pin
>>> import network
>>>
>>> wifi_en = Pin(8, Pin.OUT)
>>> wifi_en.value(1)
>>> wifi.start(tx=7, rx=6, ssid="SSID", password="PASS", baudrate=115200, wait=True)
M (239179352) [WIFI]: Initialize WiFi UART
M (239183180) [WIFI]: UART #0: initialize uart hardware
M (239213350) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=115205
M (239238364) [WIFI]: WiFi TASK STARTED
M (239341419) [ATCMD]: Check and set device baudrate
W (239981778) [ATCMD]: Trying at 230400 bd
W (240671162) [ATCMD]: Trying at 460800 bd
W (241360551) [ATCMD]: Trying at 921600 bd
W (242049936) [ATCMD]: Trying at 1000000 bd
W (242739325) [ATCMD]: Trying at 2000000 bd
E (243428705) [ATCMD]: Error setting baudrate
M (243432812) [WIFI]: UART #0 deinitialized
M (243437477) [WIFI]: WiFi TASK TERMINATED
False
```

CorruptName, 2019-12-21:
You think the flash may have destroyed the ESP8285?

loboris, 2019-12-21:
It is impossible to destroy ESP8285 by flashing!
But it is possible it was not flashed correctly. You can try some other flashing method.

You can try to attach usb->uart adapter (Rx) to esp8285 Tx (K210 Rx) without connecting adapter's Tx, to see if there is any communicattion, there should be. I think ESP8285 prints some messages after reset (at 115200 bd).

CorruptName, 2019-12-21:
I know it is no longer labeled "DanDock" like the one from the wifi flash readme.md picture. Instead it says "Maix Dock". From our previous problems, it seems they have used slower flash, what else did they change? (rhetorical)

CorruptName, 2019-12-21:
> It is impossible to destroy ESP8285 by flashing!
> But it is possible it was not flashed correctly. You can try some other flashing method.
>
> You can try to attach usb->uart adapter (Rx) to esp8285 Tx [...]

I did try this, I'll do it again and post results.

CorruptName, 2019-12-21:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40108000, len 3056, room 16
tail 0
chksum 0xd7
load 0x3fffb000, len 992, room 8
tail 8
chksum 0xd7
csum 0xd7▒"xn▒▒9▒9▒9▒▒9▒9▒▒9▒   H~▒▒V$5u▒▒▒I▒ŨR@+       ▒[▒9▒▒9▒9▒▒9▒9▒9▒▒!▒▒xZr%r)Q*!▒̙▒b▒c▒▒$v▒▒▒▒-▒▒F▒ޛ
                         "▒▒▒jq+ͫQ▒▒▒ը▒▒٨vJa▒U▒▒-!1▒▒▒▒r▒zY▒▒▒Y▒v▒       ▒r@▒y(1▒1ƭ8▒ "@j▒l

```

CorruptName, 2019-12-21:
BTW, I tried flashing back to original fw using "flash_download_tools_v3.6.8" from espressif. Didn't seem to make any difference so i went back to esptool.

CorruptName, 2019-12-21:
It also prints this when i go from LOW to HIGH on pin 8 (wifi_en):
 ```
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40108000, len 3056, room 16
tail 0
chksum 0xd7
load 0x3fffb000, len 992, room 8
tail 8
chksum 0xd7
csum 0xd7▒x▒9▒9▒9▒▒9▒9▒9▒▒9▒    H~▒▒V$5u▒▒▒I▒ըR@+       ▒[˹▒9▒9▒9▒▒9▒9▒▒9c▒!▒hxZ%r+Q*!▒̙▒b▒c▒▒▒▒vm▒▒q-▒▒▒▒ޛ
                          "▒▒▒jq+ͫ▒▒▒▒ը▒m▒(vJa▒▒U▒▒-!▒1▒▒r▒z٨▒▒Y(v▒      ▒r@p▒(1▒1▒-8▒ "▒j▒▒

```

CorruptName, 2019-12-21:
Okay, so i watched while it does "Check and set device baudrate" and I get unknown chars. I have no idea what baud it is trying to use. I have tried many possible baud rates and nothing seems to be correct. LMK what you think.

loboris, 2019-12-21:
With my firmware flashed, you should get the following messages after esp8285 reset (pin8_lo -> pin8_hi):
```
����
==============================
LoBo ESP8266 Bootloader v1.2.0
==============================
   Flash map: 2, 1MB (512+512)
  Flash mode: DOUT, [ESP8285]
Reset reason: POWER_ON

Loading configured firmware (0)
  Address: 001000
Starting firmware 0, map 2 from 001000...

ready
AT

OK
ATE0

OK
+TCPCOMMANDS:1

OK
+CWMODE:1

OK
No AP

OK
No AP

OK
WIFI DISCONNECT
+CWJAP:3

FAIL
```
The massages after `Starting firmware 0, map 2 from 001000...` will be wisible only if ESP8285 baud rate was set to 11520 (after that message ESP8285 switches to the stored AT-command baud rate).

The baud rates tried by wifi driver are 115200, 230400, 460800, 921600, 1000000, 2000000

Before starting wifi you can connect usb->uart TX line to ESP8285 and execute AT commands directly from some terminal emulator (minicom, putty)

MicroPython
```python
>>> wifi_en.value(0)
>>> wifi_en.value(1)
```

terminal:
```console
����
==============================
LoBo ESP8266 Bootloader v1.2.0
==============================
   Flash map: 2, 1MB (512+512)
  Flash mode: DOUT, [ESP8285]
Reset reason: POWER_ON

Loading configured firmware (0)
  Address: 001000
Starting firmware 0, map 2 from 001000...

ready
WIFI DISCONNECT
at

OK
at+GMR
AT version:1.7.0.0(Apr 16 2019 15:31:04)
SDK version:3.1.0-dev(3b41fcf)
Compile time: Jun 30 2019 14:56:44
ESP8266_AT_LoBo v1.3.1
OK
```
Remember that at commands must be terminated with `\r\n`, you can use ctrl+m, ctrl+j if the terminal emulator doesn't emit both on `return` key.

If you don't get the response try to change terminal emulator baud rate until you get it.

CorruptName, 2019-12-21:
> ����
> ==============================
> LoBo ESP8266 Bootloader v1.2.0
> ==============================
>    Flash map: 2, 1MB (512+512)
>   Flash mode: DOUT, [ESP8285]
> Reset reason: POWER_ON
>
> Loading [...]

Yea, I'm not getting anything like that at 115200. I get what I posted. I just reflashed your firmware again too. Seems to be going perfectly. I have tried to communicate at many different baud rates, including the ones listed with no success. Even tried using Arduino IDE serial monitor since i know it can send both LF + CR. When i send "AT" at 115200, i get a response of "!".

CorruptName, 2019-12-21:
This is what i get when booting or when toggle wifi_en.value  0 > 1
```
14:33:55.651 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
14:33:55.651 -> 
14:33:55.651 -> load 0x40108000, len 3056, room 16 
14:33:55.651 -> tail 0
14:33:55.651 -> chksum 0xd7
14:33:55.651 -> load 0x3fffb000, len 992, room 8 
14:33:55.651 -> tail 8
14:33:55.651 -> chksum 0xd7
14:33:55.651 -> csum 0xd7⸮x⸮⸮9⸮9⸮⸮9⸮9⸮9⸮⸮9⸮	H~⸮⸮V$5u⸮⸮⸮⸮I⸮ըR@+	⸮[˹⸮9⸮9⸮9⸮⸮9⸮9⸮⸮c⸮!⸮hxZr⸮⸮QJ`J⸮%"⸮⸮
%r+Q*!⸮̙⸮b⸮c⸮⸮⸮⸮vm⸮⸮q-⸮⸮⸮⸮ޛ
14:33:55.685 -> "⸮⸮Ijq+ͫ⸮⸮⸮⸮ը⸮m⸮(vJa⸮⸮U⸮⸮-!⸮1⸮⸮r⸮zY⸮⸮⸮٨v⸮	⸮r@⸮y(1⸮1֭8⸮ "@j⸮⸮ 
```

CorruptName, 2019-12-21:
I just ordered a new board, maybe i can read the flash from it and flash this one back to stock? IDK.

CorruptName, 2019-12-22:
Hey, i just realized something.. The "ESP DOWNLOAD TOOL" reports that the ESP8285 has:
```
flash vendor:
51h : N/A
flash devID:
4014h
QUAD;8Mbit
crystal:
40 Mhz
```
Could this be the issue? 8mbit flash instead of 1mbit?

loboris, 2019-12-22:
ESP8285 has 1MB Flash
8Mbit === 1Mbyte, it's ok

CorruptName, 2019-12-22:
i feel stupid, just realized that. Somehow i just did esptool read_flash for to 0x800000 tho lol.

CorruptName, 2019-12-22:
Okay, it works again on stock firmware. Just successfully read flash from the new board and wrote it to the old one. Now WiFi works again.

CorruptName, 2019-12-22:
So to recap, I don't think it's something wrong with your frimware, since the frimware posted by Sipeed doesn't work either. I'm going to post this firmware I just read here in case anyone else needs to flash back to stock and finds there is no working firmware available.

CorruptName, 2019-12-24:
So now I'm back to stock and wifi works again. I tried flashing the esp8285 with your firmware again and had the same problem. Your firmware simply does not work with my board. When I try to use stock esp8285 firmware with your K210 firmware, I get this message. Any Suggestions? Any ideas to make your firmware work with esp8285?
```
M (27406130) [WIFI]: Initialize WiFi UART
M (27409871) [WIFI]: UART #0: initialize uart hardware
M (27439922) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=115205
M (27464936) [WIFI]: WiFi TASK STARTED
M (27567998) [ATCMD]: Check and set device baudrate
M (27584015) [ATCMD]: Device communicating at 115200 bd
M (27588974) [WIFI]: WiFi device initialization start
M (28104333) [ATCMD]: AT COMMAND: [AT\r\n] Expecting: [\r\nOK\r\n]
M (28112338) [ATCMD]: AT RESPONSE: match condition 1 at position 5 (2 ms, 0 byte(s), buf_start=0)
M (28130344) [ATCMD]: AT COMMAND: [ATE0\r\n] Expecting: [\r\nOK\r\n]
M (28138349) [ATCMD]: AT RESPONSE: match condition 1 at position 7 (2 ms, 0 byte(s), buf_start=0)
M (28156355) [ATCMD]: AT COMMAND: [AT+TCPCLOSE?\r\n] Expecting: [\r\nOK\r\n]
W (28265411) [ATCMD]: AT COMMAND: TIMEOUT (100 ms, 9)
M (28379481) [ATCMD]: AT COMMAND: [AT+TCPCLOSE?\r\n] Expecting: [\r\nOK\r\n]
W (28488534) [ATCMD]: AT COMMAND: TIMEOUT (100 ms, 9)
E (28592597) [WIFI]: Firmware supporting TCP commands must be used
M (28598526) [WIFI]: UART #0 deinitialized
M (28603106) [WIFI]: WiFi TASK TERMINATED
```

loboris, 2019-12-26:
It looks there is some issue with WiFi firmware provided in this repository's `firmware/wifi/upgrade/` directory.
Could you, please, try to flash the firmware from [ESP8266_AT_LoBo](https://github.com/loboris/ESP8266_AT_LoBo).
Just clone the repository, change the working directory to `at_lobo` inside the cloned directory and flash. The flash procedure (and flash script) is the same.

If the WiFi task is not started after the first init (`WiFi TASK TERMINATED`), just try again with the correct wifi credentials (after the first try some default configuration values are saved).

CorruptName, 2019-12-26:
> It looks there is some issue with WiFi firmware provided in this repository's `firmware/wifi/upgrade/` directory.
> Could you, please, try to flash the firmware from [...]

I'm about to try. I don't have high hopes though since the firmware from sipeed didn't work either, i doubt if it's a problem with both firmwares. More likely (to me) that there was a change in hardware that requires a _different_ firmware which is causing the issue.

loboris, 2019-12-26:
I'm quite sure it will work as I've tested it today on a new M1W module, not previously flashed...

CorruptName, 2019-12-26:
```
>>> import network
>>> wifi = network.wifi
>>> wifi.debug(True)
True
>>> wifi.start(tx=7,rx=6,ssid=_ssid_,password=_pw_,wait=True)
M (185704869) [WIFI]: Initialize WiFi UART
M (185708696) [WIFI]: UART #0: initialize uart hardware
M (185738755) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=115205
M (185763769) [WIFI]: WiFi TASK STARTED
M (185866825) [ATCMD]: Check and set device baudrate
W (186507184) [ATCMD]: Trying at 230400 bd
W (187196572) [ATCMD]: Trying at 460800 bd
W (187885969) [ATCMD]: Trying at 921600 bd
W (188575360) [ATCMD]: Trying at 1000000 bd
W (189264750) [ATCMD]: Trying at 2000000 bd
E (189954141) [ATCMD]: Error setting baudrate
M (189958248) [WIFI]: UART #0 deinitialized
M (189962915) [WIFI]: WiFi TASK TERMINATED
False
>>> wifi.start(tx=7,rx=6,ssid=_ssid_,password=_pw_,wait=True)
M (202239698) [WIFI]: Initialize WiFi UART
M (202243524) [WIFI]: UART #0: initialize uart hardware
M (202273696) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=115205
M (202298711) [WIFI]: WiFi TASK STARTED
M (202401767) [ATCMD]: Check and set device baudrate
W (203042162) [ATCMD]: Trying at 230400 bd
W (203731583) [ATCMD]: Trying at 460800 bd
W (204420965) [ATCMD]: Trying at 921600 bd
W (205110362) [ATCMD]: Trying at 1000000 bd
W (205799772) [ATCMD]: Trying at 2000000 bd
E (206489176) [ATCMD]: Error setting baudrate
M (206493283) [WIFI]: UART #0 deinitialized
M (206497950) [WIFI]: WiFi TASK TERMINATED
False
>>>
```

CorruptName, 2019-12-26:
> I'm quite sure it will work as I've tested it today on a new M1W module, not previously flashed...

Does this firmware work on your M1W?
http://dl.sipeed.com/MAIX/factory_firmware/esp8266/WiFiSPIESP.ino.esp8285_40M_freq_1M_DOUT.bin

CorruptName, 2019-12-26:
> I'm quite sure it will work as I've tested it today on a new M1W module, not previously flashed...

Do your modules look like this?
[image]
[image]

CorruptName, 2019-12-26:
The only reason I say the hardware is different is because we had to decrease the flash speed for them to work with your K210 FW so I have a hunch that the flash isn't the only thing they changed with these newer modules.

loboris, 2019-12-26:
Have you erased the ESP8284 Flash first, than flashed it from inside the ESP8266_AT_LoBo repository?
I've tried on another M1W module and it also works.

The standard ESP8266/ESP8285 **AT_Firmware** can comunicate with the WiFi driver (you tested it yourself), but will not work as it does not have some new (required) TCT commands...

My board is an older one, marked "Dan Dock", but I think it is the same one.
I've also testeted on two bare M1W modules (ordered about a month ago) and it also work.

CorruptName, 2019-12-26:
> Have you erased the ESP8284 Flash first, than flashed it from inside the ESP8266_AT_LoBo repository?
> I've tried on another M1W module and it also works.
>
> The standard ESP8266/ESP8285 **AT_Firmware** [...]

Yes, i did exactly as directed.
Flash went perfectly.
Only difference is that I added "-ne" to the second step as not to erase twice.
And yes, you are correct. When i flash the "Stock firmware" (Not from their website, as the firmware on their website does not work on these modules) It will communicate with your firmware.

loboris, 2019-12-26:
As far as I know, there is a new M1W version which can also use SPI interface for communication with ESP8285, but I don't think it is available yet.

CorruptName, 2019-12-26:
https://www.amazon.com/gp/product/B07SKFLYXJ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

loboris, 2019-12-26:
Well, for now it remains a mystery.
I'll try to get some more info abot possibe hardware differences.

BTW, I've also tested with MAIX-BIT with an external ESP8285 ([ESP-01M](https://www.electrodragon.com/product/esp-01m-wifi-module-esp8285-1mb-vertical-design/)) and it works perfectly.

CorruptName, 2019-12-26:
> Well, for now it remains a mystery.
> I'll try to get some more info abot possibe hardware differences.
> 
> BTW, I've also tested with MAIX-BIT with an external ESP8285 ([ESP-01M](https://www.electrodragon.com/product/esp-01m-wifi-module-esp8285-1mb-vertical-design/)) and it works perfectly.

That's good to know. I don't think it's the fault of your firmware. I think they are doing something different with these new boards (Such as SPI) may be able to look at their source and see. But SPI doesn't make sense because of this:
```
M (27406130) [WIFI]: Initialize WiFi UART
M (27409871) [WIFI]: UART #0: initialize uart hardware
M (27439922) [WIFI]: UART #0: initialized, tx=7, rx=6, bdr=115205
M (27464936) [WIFI]: WiFi TASK STARTED
M (27567998) [ATCMD]: Check and set device baudrate
M (27584015) [ATCMD]: Device communicating at 115200 bd
M (27588974) [WIFI]: WiFi device initialization start
M (28104333) [ATCMD]: AT COMMAND: [AT\r\n] Expecting: [\r\nOK\r\n]
M (28112338) [ATCMD]: AT RESPONSE: match condition 1 at position 5 (2 ms, 0 byte(s), buf_start=0)
M (28130344) [ATCMD]: AT COMMAND: [ATE0\r\n] Expecting: [\r\nOK\r\n]
M (28138349) [ATCMD]: AT RESPONSE: match condition 1 at position 7 (2 ms, 0 byte(s), buf_start=0)
M (28156355) [ATCMD]: AT COMMAND: [AT+TCPCLOSE?\r\n] Expecting: [\r\nOK\r\n]
W (28265411) [ATCMD]: AT COMMAND: TIMEOUT (100 ms, 9)
M (28379481) [ATCMD]: AT COMMAND: [AT+TCPCLOSE?\r\n] Expecting: [\r\nOK\r\n]
W (28488534) [ATCMD]: AT COMMAND: TIMEOUT (100 ms, 9)
E (28592597) [WIFI]: Firmware supporting TCP commands must be used
M (28598526) [WIFI]: UART #0 deinitialized
M (28603106) [WIFI]: WiFi TASK TERMINATED
```
If it were SPI, we sould not be able to communicate via UART.
Maybe it's possible they are using a new variant of the esp8285? IDK, I'm probably more confused than you are.

lypanov, 2020-01-02:
@CorruptName Looks like I have the same module as you. Did you have any luck getting Wifi working with the loboris port in the end?

I need the FTP module from Loboris to get a productive workflow (Background: I have a python filesystem watcher running on my PC which uploads any file modifications via FTP to my existing ESP32 w/SPRAM via the FTP module, this plus a "module reload" workflow allowing me to do live incremental development via the REPL).

Is there any complete instructions for getting the firmware flashed as of now? I've been able to piece together some information, but I still don't understand how the small round connector in the cutout corner of the main K210 block is meant to be connected to or what it does. It also doesn't look like the Maix Dock is very breadboard friendly so I assume I have to directly solder the USB2UART to it?

Hope you can help. This was my Christmas present to myself...

CorruptName, 2020-01-02:
Unfortunately, there is no solution (yet) if you have the same module as me. You are welcome to try if you wish.

Fist off, were you able to get the loboris port of the k210 firmware working? If not, I have compiled a version with a slower flash speed that may work for you.

If you want to try and flash loboris wifi firmware, you will need to hold a jumper wire from ground to that corner pin you described while you power it up. There are several grounds to use. If you haven't soldered pin headers to the board, you should consider it. If not, you may have a hard time holding everything in place. The rest of the information you need should be available in the README found in the flash directory. But you should probably use "my version" of the instructions which address a couple issues (typos) in the loboris one.

lypanov, 2020-01-02:
No I haven't tried anything just yet. Up to now have just been reading exactly what I need to buy. Does the k210 port require the 8285 firmware update as I assumed or can it run independently from it?

CorruptName, 2020-01-02:
You can run it without. You just won't have wifi. It is possible to access wifi using UART and AT commands, though I doubt you will find any support in doing that. Also be aware, there is currently no support for the KPU. Does your board say "Dan Dock" or just "Dock"?

lypanov, 2020-01-03:
My board looks *exactly* like yours. There seems to be more than just a few versions, I've seen two different layouts of even the Maix Dock "metal covered set of chips" (sorry, not sure how to refer to the big block of chips with the cover).

Alas my use case requires wifi. I'll wait until instructions are more complete and your issues are resolved. Thanks for the help!

CorruptName, 2020-01-03:
Sure thing. When I/we find a solution, I will update here or on my fork. Unfortunately, we are stuck with stock firmware (If we need WiFi) which is severely lacking in support & documentation, which sucks because these boards are very cool with a ton of potential. If you do happen to try flashing and lose WiFi completely, i have posted a link my dump of the original WiFi firmware. If you have any other issues regarding the flashing process I can probably help you out. BTW, I would be willing to bet the K210 firmware from here won't work on your board either without making changes and recompiling. We already went through that on another (Now Closed) issue. I can post that "modified" firmware if anyone needs it. Basically it just has slower access rate to the flash as it seems the new boards use slower flash.

If I were you I would try and see what happens. Nothing to lose and just because our boards look alike, does not mean they are the same.

And BTW: I think that's called a "Lid" but i'm not 100%.

bkircher67, 2020-03-07:
> This is what i get when booting or when toggle wifi_en.value 0 > 1
>
> ```
> 14:33:55.651 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
> 14:33:55.651 ->
> 14:33:55.651 -> load 0x40108000, len 3056, room [...]

Hello,

either we both have corrupted boards - or there is an issue with the latest firmware as I get the same results
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40108000, len 3056, room 16 
tail 0
chksum 0xd7
load 0x3fffb000, len 992, room 8 
tail 8
chksum 0xd7
csum 0xd7�"x�9��9�9�9�9�9�9�9�  H~���$5�Q�I�p=B5@"x�9�9�9�9�9�9�9��9�   �(-I-Ajp��@�(X%�5a!!�hx)Q*!����`5g!ƛt5$�q-"�1"8��Ijq+%qkY.vkA�r�P@� (����(1�1   �r�zY�z����@jp5-�~11-p"8�"@j�N

Initially with the sipeed MicroPython and the original ESP8285 software It worked ...

CorruptName, 2020-03-08:
> Hello,
>
> either we both have corrupted boards - or there is an issue with the latest firmware as I get the same results
> ets Jan 8 2013,rst cause:1, boot mode:(3,7)
>
> load 0x40108000, len 3056, room 16 [...]

If you need to, you can load the firmware that I linked above to get your wifi working again.

poysh, 2020-03-13:
Just ran into the same issue: added bonus content, when I flash the ESP with the `./flash.sh -t 1MB -m dout` after erasing it, I get the following outputs

```
Hash of data verified.
Flash params set to 0x0320
Compressed 4080 bytes to 2696...
Wrote 4080 bytes (2696 compressed) at 0x00000000 in 0.1 seconds (effective 509.9 kbit/s)...
Hash of data verified.
Compressed 1684 bytes to 1206...
Wrote 1684 bytes (1206 compressed) at 0x0007b000 in 0.0 seconds (effective 420.9 kbit/s)...
Hash of data verified.
Compressed 435876 bytes to 313593...
Wrote 435876 bytes (313593 compressed) at 0x00001000 in 7.8 seconds (effective 447.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

A fatal error occurred: Timed out waiting for packet header
```
or sometimes I would get 
```
Staying in bootloader.
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 3201, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 3194, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2960, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
  File "/usr/local/bin/esptool.py", line 977, in soft_reset
    self.flash_finish(True)
  File "/usr/local/bin/esptool.py", line 596, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "/usr/local/bin/esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/usr/local/bin/esptool.py", line 347, in command
    p = self.read()
  File "/usr/local/bin/esptool.py", line 292, in read
    return next(self._slip_reader)
  File "/usr/local/bin/esptool.py", line 2041, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/home/liph/.local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Flash write failed!
``` 

If I start the board afterwards and try to connect to a WIFI I run into the same issue as described before with the baudrates

Hardware: Maix Dock with ota firmware

andriyadi, 2020-05-15:
> Hello,
>
> either we both have corrupted boards - or there is an issue with the latest firmware as I get the same results
> ets Jan 8 2013,rst cause:1, boot mode:(3,7)
>
> load 0x40108000, len 3056, room 16 [...]

I experience the same problem. I'm using Maix Go which has M1W module, that I purchased few weeks ago.

I think I may have a solution. So, after back and forth erasing, flashing with LOBO AT firmware, I decided to flash with original firmware (that luckily I dumped before doing anything).

Then, to flash LOBO AT firmware, I don't flash the whole thing, instead just the application, by this command:
**`./flash.sh -t 1MB -m dout -fo`**

Notice there's **`-fo`** parameter

Then, when after rebooting the board, I saw this serial log from esp8285:
```
ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 1856, room 16 
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8 
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8 
tail 0
chksum 0x79
csum 0x79

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

syÛ—ïµÅparam error, use last saved param!
˛
ready
```

Notice there's: 
```
syÛ—ïµÅparam error, use last saved param!
```
Which I think it's OK, because I don't see it again after next reboot.

Then I confirm that I have LOBO AT Firmware by typing: `AT+GMR` and got this:

```
AT version:1.7.0.0(Apr 16 2019 15:31:04)
SDK version:3.1.0-dev(3b41fcf)
Compile time: Dec 26 2019 13:43:42
ESP8266_AT_LoBo v1.3.1
OK
```

I hope it may help someone. Good luck!

faisyl, 2021-06-02:
I just went through all of @CorruptName did. I saw the exact same problem with my Maix M1 dock board. But flashing only the application after restoring factory firmware as @andriyadi suggests, fixes it for me as well.