loboris/MicroPython_K210_LoBo #20: FILESIZE incorrect in BUILD.sh on MacOS

derekmulcahy, 2020-02-09:
The code for computing the FILESIZE in BUILD.sh is incorrect on MacOS.

`FILESIZE=$(stat -s MicroPython.bin | cut -d' ' -s -f 1 | cut -d'=' -s -f 2)`

is getting the value of the first field in the output of stat -s and this is st_dev=16777223 on my machine. Alternative implementations are

`FILESIZE=$(eval $(stat -s MicroPython.bin); echo $st_size)`

or 
 
`FILESIZE=$(stat -s MicroPython.bin | cut -d' ' -s -f 8 | cut -d'=' -s -f 2)`

which also works for me but I am not sure if this is always true.

This code appears twice in BUILD.sh and both need updating.

Thanks for all the updates!

derekmulcahy, 2020-02-09:
Another suggestion for MacOS. The current build process uses features of Make which are not available in the default Mojave MacOS version of make(3.81). Adding

`PATH="/usr/local/opt/make/libexec/gnubin:$PATH"`

to the MacOS specific flow through BUILD.sh will allow the later homebrew versions of make(4.3) to be used if installed.

Here is the output I get when I don't use make from home-brew

```
====[ BUILD: Building on Darwin skye.local 18.7.0 Darwin Kernel Version 18.7.0: Thu Jan 23 06:52:12 PST 2020; root:xnu-4903.278.25~1/RELEASE_X86_64 x86_64
=====[ BUILD: File system image created
=====[ BUILD: 3 MB firmware

=====[ BUILD: UPDATE mk files
../../micropython/py/mkrules.mk:94: target `build/./' given more than once in the same rule.
../../micropython/py/mkrules.mk:94: target `build/./' given more than once in the same rule.
../../micropython/py/mkrules.mk:94: target `build/./' given more than once in the same rule.
../../micropython/py/mkrules.mk:94: target `build/./' given more than once in the same rule.

===========================================
=====[ BUILD: BUILDING MicroPython FIRMWARE
===========================================
make: source: Command not found
make: source: No such file or directory
make: *** [do_mk] Error 1
====================
==== Build ERROR ===
=
===================
```
Cheers,
Derek

derekmulcahy, 2020-02-09:
Sorry for the chain of comments. CLEAN.sh also requires a newer version of make.

loboris, 2020-02-10:
Thank you for your comments.
I'll test your suggestions when I'll find some time. I'm not using MacOS regularly (I'm only testing in a virtual machine) so any suggestion from experienced  Mac user is welcome.

jetpax, 2020-03-03:
In order to use  brew's newer version of make, I needed to export 
PATH="$(brew --prefix)/opt/make/libexec/gnubin:$PATH"
instead of PATH="/usr/local/opt/make/libexec/gnubin:$PATH"