loboris/MicroPython_K210_LoBo #21: Can't use SSL with latest firmware

kartben, 2020-02-19:
My ESP8285 firmware is up-to-date, and I can't seem to use SSL with the latest firmware as the AT firmware crashes. Please see below:

```python
[INFO] COM Port Selected Manually:  /dev/tty.usbserial-14401 

--[ MicroPython terminal  ver. 5.1.4 ]-- 
--[ Press ESC twice for command mode ]-- 

>>> import network, machine, os
>>> wifi = network.wifi
>>> wifi.start(tx=7, rx=6, ssid="***REDACTED***", password="***REDACTED***", wait=True)
True
>>> 
>>> wifi.ifconfig()
('192.168.1.48', '192.168.1.1', '255.255.255.0', '192.168.1.1', '208.67.222.222')
>>> 
>>> requests = network.requests
>>> requests.debug(True)
>>> 
>>> res = requests.get('http://loboris.eu/K210/test.txt')
M (22380713) HTTP_CLIENT: http_on_headers_complete, status=200, offset=269, nread=269
>>> res[0]
200
>>> res2 = requests.get('https://loboris.eu/K210/test.txt')
E (30956294) TRANS_SSL: Write error, errno=Network is unreachable
E (30962116) HTTP_CLIENT: Error write request
E (33013365) [REQUESTS]: HTTP Request failed: 28675 []
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: HTTP Request failed
>>> res2[0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> 
```

Logs from the ESP UART:
```
==============================
LoBo ESP8266 Bootloader v1.2.0
==============================
   Flash map: 2, 1MB (512+512)
  Flash mode: DOUT, [ESP8285]
Reset reason: EXCEPTION (0)

Loading configured firmware (0)
  Address: 001000
Starting firmware 0, map 2 from 001000...

ready
WIFI DISCONNECT
WIFI CONNECTED
WIFI GOT IP
AT

OK
AT

OK
ATE0

OK
+TCPCOMMANDS:1

OK
+CWMODE:1

OK
+CWJAP:"***REDACTED***","ac:84:***REDACTED***",11,-63,0

OK

OK

OK
+CIPRECVMODE:0

OK

OK

OK
+CIPSNTPCFG:0

OK

OK
+CIPSNTPCFG:1,0,"pool.ntp.org"

OK
+CIPSTA:ip:"192.168.1.48"
+CIPSTA:gateway:"192.168.1.1"
+CIPSTA:netmask:"255.255.255.0"

OK
+CIPDNS_CUR:192.168.1.1
+CIPDNS_CUR:208.67.222.222

OK
+CIPSSLSIZE:4096

OK

OK
0,CONNECT

OK

>
+TCPSEND:109

+TCPSEND:Complete

OK

+TCP,0,9,1331:HTTP/1.1 200 OK
Date: Wed, 19 Feb 2020 17:14:58 GMT
Server: Apache/2.4.29 (Ubuntu)
Last-Modified: Wed, 20 Mar 2019 13:08:43 GMT
ETag: "426-5848652314cc0"
Accept-Ranges: bytes
Content-Length: 1062
Vary: Accept-Encoding
Content-Type: text/plain; charset=UTF-8

===============
Welcome to K210
===============

General Specifications:
-----------------------
 * The K210 includes two 64-bit RISC-V CPU cores, each with a built-in independent FPU.
 * KPU high performance Convolutional Neural Network (CNN) hardware accelerator.
   The primary functions of the K210 are machine vision and hearing,
   which includes the KPU for computing convolutional neural networks and
   an APU for processing microphone array inputs.
 * The K210 features a Fast Fourier Transform (FFT) Accelerator for high performance
   complex FFT calculations. As a result, for most machine learning algorithms,
   the K210 has high-performance processing power.
 * Advanced TSMC 28nm process, temperature range -40°C to 125°C
 * Firmware encryption support
 * Unique programmable IO array maximises design flexibility
 * Low voltage, reduced power consumption compared to other systems
   with the same processing power
 * 3.3V/1.8V dual voltage IO support eliminates need for level shifters

---------
2019 LoBo
---------

OK
0,CLOSED
+CIPSSLSIZE:16384

OK
0,CONNECT

OK

>
+TCPSEND:109

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40108000, len 3056, room 16 
tail 0
chksum 0xd7
load 0x3fffb000, len 992, room 8 
tail 8
chksum 0xd7
csum 0xd7�
==============================
LoBo ESP8266 Bootloader v1.2.0
==============================
   Flash map: 2, 1MB (512+512)
  Flash mode: DOUT, [ESP8285]
Reset reason: EXCEPTION (28)

Loading configured firmware (0)
  Address: 001000
Starting firmware 0, map 2 from 001000...

ready
WIFI CONNECTED
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
WIFI GOT IP
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0

ERROR
AT+TCPCLOSE=0
```

loboris, 2020-02-22:
Sorry for late reply.
I can't reproduce this issue. Does it occur every time you try the https request?

Here is my log, i've tried to repeat the the request 10 times and had no error.
Maybe you have weak WiFi signal?

In your case ESP8285 crashes with Exception 28 (A load referenced a page mapped with an attribute that does not permit loads).
I've found some references to this exceptio occuring when WiFi was disconnected and reconnected, but at this time I'm not shure how to fix it.

I'm now working on using ESP32 as WiFi interface, after I'll finish it, I'll try to check the ESP8266 firmware for possible cause of this issue.

```python
>>> res = requests.get('http://loboris.eu/K210/test.txt')
M (141056968) [ATCMD]: AT COMMAND: [AT+CIPSSLSIZE?\r\n] Expecting: [\r\nOK\r\n], [\r\nERROR\r\n]
M (141070675) [ATCMD]: AT RESPONSE: match condition 1 at position 34 (4 ms, 24 byte(s), buf_start=16)
M (141079639) [ATCMD]: AT COMMAND: [AT+CIPSSLSIZE=16384\r\n] Expecting: [\r\nOK\r\n], [\r\nERROR\r\n]
M (141092678) [ATCMD]: AT RESPONSE: match condition 1 at position 40 (1 ms, 0 byte(s), buf_start=40)
W (141101689) [WIFI_TASK]: no match, removed [39.121.8\r\n^^^^^^+CIPSSLSIZE:40]
M (141109383) [ATCMD]: AT COMMAND: [AT+TCPSTART=0,"TCP","loboris.eu",80,3,0\r\n] Expecting: [0,CONNECT\r\n], [\r\nERROR\r\n], [\r\n+TCPSTART]
M (141205714) [ATCMD]: AT RESPONSE: match condition 1 at position 16 (79 ms, 0 byte(s), buf_start=16)
W (141214679) [WIFI_TASK]: no match, removed [96\r\n^^^^^^^^^^^^^]
M (141221398) [ATCMD]: AT COMMAND: [AT+TCPSEND=0,109\r\n] Expecting: [\r\n>], [\r\n+TCPSEND:Busy\r\n]
M (141233722) [ATCMD]: AT RESPONSE: match condition 1 at position 16 (1 ms, 0 byte(s), buf_start=16)
M (141242591) [ATCMD]: AT COMMAND: [Send data] Expecting: [\r\n+TCPSEND:Complete\r\n], [\r\nOK\r\n], [\r\nERROR\r\n], [FAIL]
W (141254307) [ATCMD]: data: [GET /K210/test.txt HTTP/1.1\x0D\x0AUser-Agent: MicroPython HTTP Client/1.0\x0D\x0AHost: lobo]
M (141306748) [ATCMD]: AT RESPONSE: match condition 2 at position 35 (31 ms, 22 byte(s), buf_start=19)
M (141315807) [WIFI_TASK]: +TCP data request found [+TCP,0,9,1331:]
M (141322554) [WIFI_TASK]: Data for link_id 0 (srv=9), len=1331 [+TCP,0]
M (141329759) [WIFI_TASK]: +TCP Requesting data
M (141450805) [WIFI_TASK]: received (128 ms); socket: len=1331, ovf=0, tail=1331, head=0
Y (141458621) [WIFI_TASK]: Wait for data processing
M (141464351) HTTP_CLIENT: http_on_headers_complete, status=200, offset=269, nread=269
Q (141571843) [WIFI_TASK]: Data processed in 107 ms
M (141576622) [WIFI_TASK]: +TCP Confirm receive
Q (141582180) [WIFI]: Close socket 0 (0)
M (141586270) [ATCMD]: AT COMMAND: [AT+TCPCLOSE=0\r\n] Expecting: [\r\nOK\r\n], [\r\nERROR\r\n]
M (141599832) [ATCMD]: AT RESPONSE: match condition 1 at position 0 (3 ms, 0 byte(s), buf_start=0)
Y (141608529) [WIFI_TASK]: connection for socket with link_id 0 closed, active=286 ms, time=136 ms
W (141618151) [WIFI_TASK]: closed, removed [0,CLOSED]
Q (141623707) [WIFI_TASK]: Processing delay
>>> Q (141677855) [WIFI_TASK]: Continue

>>> res[0]
200
>>> res2 = requests.get('https://loboris.eu/K210/test.txt')
M (177172575) [ATCMD]: AT COMMAND: [AT+CIPSSLSIZE?\r\n] Expecting: [\r\nOK\r\n], [\r\nERROR\r\n]
M (177185807) [ATCMD]: AT RESPONSE: match condition 1 at position 35 (3 ms, 25 byte(s), buf_start=16)
W (177194823) [WIFI_TASK]: no match, removed [^^^^^^^^^^^^^^\r\n+CIPSSLSI]
M (177202169) [ATCMD]: AT COMMAND: [AT+TCPSTART=0,"SSL","loboris.eu",443,3,0\r\n] Expecting: [0,CONNECT\r\n], [\r\nERROR\r\n], [\r\n+TCPSTART]
M (177596935) [ATCMD]: AT RESPONSE: match condition 1 at position 16 (378 ms, 0 byte(s), buf_start=16)
W (177605988) [WIFI_TASK]: no match, removed [ZE:16384\r\n^^^^^^^]
M (177612706) [ATCMD]: AT COMMAND: [AT+TCPSEND=0,109\r\n] Expecting: [\r\n>], [\r\n+TCPSEND:Busy\r\n]
M (177624943) [ATCMD]: AT RESPONSE: match condition 1 at position 16 (1 ms, 0 byte(s), buf_start=16)
M (177633812) [ATCMD]: AT COMMAND: [Send data] Expecting: [\r\n+TCPSEND:Complete\r\n], [\r\nOK\r\n], [\r\nERROR\r\n], [FAIL]
W (177645528) [ATCMD]: data: [GET /K210/test.txt HTTP/1.1\x0D\x0AUser-Agent: MicroPython HTTP Client/1.0\x0D\x0AHost: lobo]
M (177709971) [ATCMD]: AT RESPONSE: match condition 2 at position 35 (43 ms, 22 byte(s), buf_start=19)
W (177719024) [WIFI_TASK]: no match, removed [^^^^^^^^^^\r\nOK\r\n^^^\r\n+TCP]
W (177747794) [WIFI_TASK]: no match, removed [S]
Q (177752139) [WIFI_TASK]: more data, check again
M (177757531) [WIFI_TASK]: +TCP data request found [+TCP,0,9,1331:]
M (177764294) [WIFI_TASK]: Data for link_id 0 (srv=9), len=1331 [+TCP,0]
M (177771498) [WIFI_TASK]: +TCP Requesting data
M (177892556) [WIFI_TASK]: received (128 ms); socket: len=1331, ovf=0, tail=1331, head=0
Y (177900373) [WIFI_TASK]: Wait for data processing
M (177906101) HTTP_CLIENT: http_on_headers_complete, status=200, offset=269, nread=269
Q (178014081) [WIFI_TASK]: Data processed in 109 ms
M (178018860) [WIFI_TASK]: +TCP Confirm receive
Q (178024418) [WIFI]: Close socket 0 (0)
M (178028508) [ATCMD]: AT COMMAND: [AT+TCPCLOSE=0\r\n] Expecting: [\r\nOK\r\n], [\r\nERROR\r\n]
M (178042069) [ATCMD]: AT RESPONSE: match condition 1 at position 0 (4 ms, 0 byte(s), buf_start=0)
[... 11 lines cut ...]
===============

General Specifications:
-----------------------
 * The K210 includes two 64-bit RISC-V CPU cores, each with a built-in independent FPU.
 * KPU high performance Convolutional Neural Network (CNN) hardware accelerator.
[... cut ...]
   complex FFT calculations. As a result, for most machine learning algorithms,
   the K210 has high-performance processing power.
 * Advanced TSMC 28nm process, temperature range -40°C to 125°C
 * Firmware encryption support
 * Unique programmable IO array maximises design flexibility
 * Low voltage, reduced power consumption compared to other systems
   with the same processing power
 * 3.3V/1.8V dual voltage IO support eliminates need for level shifters

---------
2019 LoBo
---------

>>> 
```

kartben, 2020-02-27:
Thank you for the feedback, Boris. This may have to do with the ESP8285 indeed. 
I will try to investigate further. In the meantime, if that's OK I suggest we leave this issue open?