loboris/MicroPython_K210_LoBo #22: machine.UART not working

poysh, 2020-03-11:
In every Version I tested to far of the prebuild images I can't initialise a UART. 
The board just resets. 

`>>> import machine
>>> uart = machine.UART(1, tx=21, rx=20)
>>> core 0, core dump: misaligned load
E (65958383) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (65975134) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (65991887) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66008641) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66025394) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66042147) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66058900) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66075653) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66092406) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66109159) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66125912) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66142665) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66159419) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66176172) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66192925) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66209678) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66226431) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66243184) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66259937) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66276690) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66293443) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66310196) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66326950) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66343703) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66360456) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66377209) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66393962) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66410715) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66427468) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66444221) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66460974) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66477727) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66494481) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66511234) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66527987) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66544740) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66561493) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66578246) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66594999) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66611752) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66628505) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66645258) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66662012) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66678765) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (66695518) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
[... cut ...]
E (68722643) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (68739396) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
E (68756150) FreeRTOS: (/home/LoBo2_Razno/K210_razvoj/MicroPython_K210_LoBo/k210-freertos/platform/sdk/kendryte-freertos-sdk/lib/freertos/tasks.c:4105) pxTCB == pxCurrentTCB[uxPsrId]
M (3655) [K210_MAIN]: Default flash configuration set`

poysh, 2020-03-11:
https://loboris.eu/forum/showthread.php?tid=1024 should be updated by tomorrow

loboris, 2020-03-12:
Update, you can test now.

poysh, 2020-03-12:
Works flawlessly, no misaligned loads, no issue so far, thank you!

closed 2020-03-12 by poysh