loboris/MicroPython_K210_LoBo #25: TFT Display on Maix Amigo

elixx, 2020-11-21:
I have the Sipheed Maix Amigo flashed and working good so far, very fast so far, but I'm trying to get all of the peripherals to work. Is there any way to mirror the screen instead of just changing orientation, or a device level config that I'm missing?

`tft.init()`

[image]

`tft.init(rot=tft.PORTRAIT)`

[image]

`tft.init(rot=tft.LANDSCAPE_FLIP)`

[image]

loboris, 2020-12-22:
Sorry for late reply.
It can be done, but it is not configurable at the moment from MicroPython.
I'll try to add the mirroring options before the next update (expected by the end of this week).

ramsesgarciad, 2024-03-23:
this error are solded?